m-lundberg/simple-pid #1: non-monotonic time issue

deniz195, 2018-10-01:
Hi martin,

i'm using a similar PID implementation on a raspberry pi in a production environement (PID code is also based Brett Beauregard's code at [](https://github.com/deniz195/jataruku.git)).

Turns out that both `time.time()` and `datetime.datetime.now()` are non-monotonic time signals. I.e. if a system service or the user turns the time back, the PID will see negative time differences. This happens on many systems (in particular the raspberry pi) regularly which has given me rare, but quite erratic behavior of the machine. 

Fixed the problem using `time.monotonic()` which is available since python 3.3. Maybe this is helpful to your code too.

Best,
Deniz

m-lundberg, 2018-10-02:
Hello,

Thanks for the feedback!

This sounds like a tricky bug to track down if you are unaware of this. I will definitely look into using `time.monotonic()` instead when I get the chance. It might save me or someone else a headache in the future.

closed 2018-10-02 by m-lundberg

m-lundberg, 2018-10-02:
I changed to using `time.monotonic()` for calculating time instead as you suggested if the python version is > 3.3. If `time.monotonic()` is not available it will fall back to using `time.time()` instead, with a warning to the user.

Thank you again for the feedback @deniz195!

commit d9c8cdd3 (2018-10-02): Use time.monotonic() for python>=3.3 to ensure no negative time deltas due to changed system time. Will default back to time.time() if time.monotonic() is not available. Fixes #1.
commit f8cc6243 (2018-10-02): Use time.monotonic() for python>=3.3 to ensure no negative time deltas due to changed system time. Will default back to time.time() if time.monotonic() is not available. Fixes #1.