m-lundberg/simple-pid #2: Tracking and output of PID terms to PV

G00364778, 2019-01-30:
I would like to track to individual contributions of the PID terms to the PV output in order to facilitate manual tuning of hard to tune loops. Can this be facilitated in the current implementation?

m-lundberg, 2019-01-30:
Hello!

This is not exactly supported at the moment but as a workaround you can get the P and I terms by doing something like this in your loop, after calling the PID update:

```python
p, i = pid._proportional, pid._error_sum
```

Unfortunately you can't get the D term as easily since it's not stored. You should be able to calculate it like this though:

```python
d = -(pid.Kd * (PV - pid._last_input))
```

(written off the top of my head so I might have made a mistake)

I might add a more convenient way of doing this though as it seems like a useful feature to have.

m-lundberg, 2019-01-30:
I added a `components`-property to the PID controller to do this, it can be used like this:

```python
while controlling:
    cv = pid(pv)
    p, i, d = pid.components  # the separate terms are now in p, i, d
```

You can install the latest version and try it out. Let me know what you think!

closed 2019-01-30 by m-lundberg

G00364778, 2019-02-07:
Thanks for the quick update and support on the product.