m-lundberg/simple-pid #11: Question: Positive/Negative outputs

bluckau, 2019-10-09:
Hi,

I would like to request clarification in the documentation on whether or not the Simple PID can be used for dual-controls (e.g. heating/cooling). If it is already set up to do that, please explain how (positive/negative output values?)

m-lundberg, 2019-10-14:
Hi,

Positive and negative output values is supported by setting the output limits to the desired range, for example

```python
pid.output_limits = (-10, 10)
```

will give output values between -10 and 10. You can also set

```python
pid.output_limits = None
```

to put no limits on the output values. Of course the output values you get also depend on the tunings Kp, Ki and Kd and your input.

If you mean if the PID can work in [reverse mode](http://brettbeauregard.com/blog/2011/04/improving-the-beginners-pid-direction/), meaning that an increase in the output leads to a decrease in the input (like when cooling), it can be achieved by setting the tunings to negative values:

```python
pid.tunings = (-1, -0.1, 0)
```

(Note that all the tunings should have the same sign).

m-lundberg, 2019-12-11:
I added some clarification on reverse mode to README.md

closed 2019-12-11 by m-lundberg