m-lundberg/simple-pid #12: restarting the PID from an interrupt

brianjmurrell, 2019-11-02:
Am I understanding correctly that if I needed to interrupt the PID and wanted to resume it from where it left off, I would need to save the:

```
        self._proportional = 0
        self._integral = 0
        self._derivative = 0
```

values and then set them when starting it back up again?  Would I also need to do something with:

```
        self._last_time = _current_time()
        self._last_output = None
        self._last_input = None
```

It might be nice to have `save()` and `load()` methods to automate this.

m-lundberg, 2019-11-13:
Hello, sorry for the slow response.

As long as you keep the same PID-object, `self._proportional` etc. will be kept so you don't need to save them. However, if you go a long time without updating the PID you will have a very large time delta the next time you call it which would lead to some weird values in the beginning. You could set

```python
pid._last_time = PID._current_time()
```

before starting again and it should work OK.

m-lundberg, 2019-12-11:
The way I see it, `save()` and `load()` methods would be unnecessary so I'm closing this. Feel free to reopen if you disagree!

closed 2019-12-11 by m-lundberg