m-lundberg/simple-pid #14: Preferred contribution method?  Augmentation and error found

aseelye, 2020-03-15:
Hi,

I couldn't find a better way to contact you, so doing this.  First, thank you for this, it's a good library that I was able to work into my project with ease.  In digging in, I found an error in the implementation of the return value as it pertains to the Ki handling.  I also implemented a Ki windup protection outside the pid.outside_limits value (that is, perhaps you don't want the final I component exceeding a value equal to that of the final return value).  If you prefer to write your own, I can point out what I did, or let me know if you just want a PR filed.  If needed, my contact is on my GH page.

Thanks!

-Aaron

m-lundberg, 2020-03-15:
Hello!

I'm glad you found the library useful! I appreciate you taking the time to report the issues you had.

If you have already implemented the changes then I would be happy to get a PR (perhaps two separate PRs if possible since there seems to be one bugfix and one new feature).

aseelye, 2020-03-15:
Sounds good, you should see two separate PRs now.

m-lundberg, 2020-03-15:
Great, thanks! I'll take a look as soon as I get a chance.

bboysteed, 2020-03-23:
why me run the example file:water_boiler.py in the command line get an unexpected result.but when i run this file in the original python IDE ,it  works.

closed 2020-09-17 by m-lundberg