m-lundberg/simple-pid #20: usage question

HaroldSP, 2020-06-09:
Hello everyone!

Is there any more detailed usage example?

Let's say, I have a car with a range sensor, which gives me the distance to the wall, and I have the desired distance, so according to an error I send different PWM values to motors.

How the code would look like?

Thanks.

m-lundberg, 2020-06-10:
Hello!

There are a few examples in the [readme](https://github.com/m-lundberg/simple-pid/blob/master/README.md) and [a complete example application](https://github.com/m-lundberg/simple-pid/tree/master/examples/water_boiler). Is there something specific you don't understand from those examples?

You probably want to set the `output_limits` on your PID controller to `(0, 255)` or whatever the max PWM output is in your case.

HaroldSP, 2020-06-17:
Thanks! 

The example helped.

closed 2020-06-17 by HaroldSP