m-lundberg/simple-pid #26: Setting last output, even when auto mode is on.

keck-in-space, 2020-10-06:
Hello,

I was working with setting the last_output value by calling ```pid.set_auto_mode(True, last_output=X)```. The issue I was having is that the auto mode was already on, so the last_output value was not being updated.

This looks to be the intended functionality as the code for ```set_auto_mode``` checks if auto_mode is currently on. What I'm wondering is if we could update the docs to state that if you want to set the last_output value, you must first turn auto_mode off, or if we could add a function solely for setting the last_output value.

It's also possible that setting the last output value during an auto mode is not advisable.

Thanks!

m-lundberg, 2020-11-13:
Hello.

I don't really know why you would want to set the last output while running the PID? It is only the value that is returned either if not enough time has passed since last calling it or if the PID is not running (not in auto mode). Could you please explain what use case you had where you needed to do this?

keck-in-space, 2020-11-13:
Well I'm not trying to advocate for having this necessarily work while the PID is running. I'm just saying that it was difficult to know when looking at the docs that it was required that I turn off ```auto_mode``` before being able to set the ```last_output``` value. It worked if I turned off auto mode first and then set the last output value, and then turned auto mode back on.

This was useful when we had a set up with a very long response time. Our PID tuning parameters were set up in such a way that the control worked great at keeping us at a target point, but had a lot of trouble getting to that point from zero (like on system startup). If we set the last output value to what we anticipated was close to the endpoint, we had better control without a lot of sway. We had several setpoints we needed to reach, so we needed to tune the output value while our set up was running.

Perhaps I just misread the docs, looking over them now it does mention that disabling the auto mode is something you can do. I just didn't realize it was something you had to do to set the output value.

m-lundberg, 2020-11-13:
I'm still not sure I understand exactly how you mean that setting the `last_output` value helped. It sounds like you wanted to manually get a higher value in the beginning to get your system close to the target? That's what turning auto mode on/off is for. You could just turn the PID off and manually feed that system the higher value until you get close enough for the PID to take over, and then enable auto mode. You could then pass the value that you manually fed to the system as the `last_output` argument of `set_auto_mode()` to avoid a big jump since the PID will then pick up from where you left off. But perhaps I'm just misunderstanding what you wanted to achieve.

Either way, I could make it clearer in the documentation that the argument only does something when going to auto mode.

keck-in-space, 2020-11-13:
Yes that's exactly what we ended up doing. I am really just saying that process wasn't super intuitive from my first reading of the documentation. Thanks for taking the time to understand!

m-lundberg, 2020-11-16:
Okay, then we are on the same page. I'm not an expert on setting up PID controllers but from my understanding it's not completely uncommon to have to perform such a strategy.

I'll try to make the behaviour a bit clearer in the documentation!

commit c44adfbb (2020-11-16): Clarify documentation on last_output argument for set_auto_mode()

Clarify in the documentation for set_auto_mode() that the last_output argument has no effect if the PID is already in auto mode when calling the method. Closes #26 .
closed 2020-11-16 by m-lundberg