m-lundberg/simple-pid #32: port to CircuitPython

mgenova79, 2021-03-04:
Not an issue, but just a question.

Is there any chance someone would be willing to port this to CircuitPython for use with adafruit boards?  There currently is no PID library available for CircuitPython (or even micropython that I can find) and I think that it would be great to have your library available to circuitpython compatible microcontrollers. I actually tried myself by using mpy cross and I successfully compiled a .mpy but the resulting library fails to create an instance with:
`from simple_pid import PID
pid = PID(1, 0.1, 0.05, setpoint=1)`

I'm sure there are many other CircuitPython users that would like to use your library as it is very easy to use, and currently would be the only go-to solution.

Thank you,
Mike G

mz3-andrey, 2021-03-17:
I use uPython at work and PID. I could do this when I'm again working on a project that requires both.

m-lundberg, 2022-04-30:
I've never used it myself but I've seen that there exists a fork for MicroPython which might help you out: https://github.com/gastmaier/micropython-simple-pid

closed 2022-04-30 by m-lundberg