m-lundberg/simple-pid #42: Is it possible to use variable inputs?

jwagn96, 2021-06-22:
Hello, i'm new to your PID-library. It works just fine with an constant input. Now I want to figure out how to make this controller work with variable inputs like ramps. Just restarting the PID in a timed loop won't work for the I-and D-values as the controller is restarted every iteration.
Is there already a possibility to feed an array of setpoints (for example a temperature curve) to the controller?
Thanks in advance
jwagn

m-lundberg, 2021-06-22:
I'm not sure I understand what you mean. Do you mean that you want to ramp up the setpoint while controlling? You can do that by simply setting `pid.setpoint` to the value you currently want the setpoint to be. You don't need to reset the whole PID to do it.

Something like this:

```python
# Create the controller once
pid = PID(1, 0.1, 0.05, setpoint=1)

# Assume we have a system we want to control in controlled_system (a heater for example)
v = controlled_system.update(0)

while True:
    # For each value, give the current value to the PID and get a control variable
    control = pid(v)
    
    # Feed the PID output to the system and get its current value
    v = controlled_system.update(control)

    # Update the setpoint
    pid.setpoint = next_setpoint()
```

Or maybe I just completely misunderstood what you meant?

jwagn96, 2021-06-22:
that's exactly what I meant. Thank you for your quick reply!

m-lundberg, 2021-06-23:
Happy to help!

closed 2021-06-23 by m-lundberg