m-lundberg/simple-pid #58: Document reverse acting control using simple_pid

kjkoster, 2022-04-30:
Dear All,

Really more of a support question, but can the documentation be enhanced to show how to use simple_pid for reverse acting control?

Kees Jan

Juasmis, 2022-09-13:
You need to use negatives gains for all the control actions that you use (proportional, integral, derivative)

I agree that maybe in the README there could be a bit more documentation about the structure of the controller and so on but since it's a classical control strategy you can find a lot of information on Google

[Wikipedia](https://en.wikipedia.org/wiki/PID_controller?wprov=sfla1),
[related question in stackiverflow](https://www.google.com/url?sa=t&source=web&rct=j&url=https://electronics.stackexchange.com/questions/453801/can-my-pid-controller-have-negative-p-value-and-positive-i-value-for-a-pid-logic&ved=2ahUKEwjey4SfppH6AhXwiP0HHcBnDYQQFnoECEcQAQ&usg=AOvVaw2RJtniBwIW9xKK7BlK256o)

Juasmis, 2022-09-19:
Checking the README it actually appears to be documented:
> To use the PID in [reverse mode](http://brettbeauregard.com/blog/2011/04/improving-the-beginners-pid-direction/), meaning that an increase in the input leads to a decrease in the output (like when cooling for example), you can set the tunings to negative values:
pid.tunings = (-1.0, -0.1, 0)
Note that all the tunings should have the same sign.

closed 2022-09-19 by kjkoster