m-lundberg/simple-pid #59: running the code and the simple provided example issue

hamedhkme, 2022-05-16:
I am going to use simple-pid algorithm. 
I have installed simple-pid, however, according to the script below I need to have a “controlled_system” that I am not sure what it is. I am wondering if you could help me with that.
 
############################################################
from simple_pid import PID
pid = PID(1, 0.1, 0.05, setpoint=1)
 
# Assume we have a system we want to control in controlled_system
v = controlled_system.update(0)
 
while True:
    # Compute new output from the PID according to the systems current value
    control = pid(v)
    
    # Feed the PID output to the system and get its current value
    v = controlled_system.update(control)
###########################################################
Also, I tried to figure out the algorithm through the water_boiler example, and for that one, I could not rerun it. 
I would appreciate it if you could help.

rnestler, 2022-07-12:
Well the controlled system is where you want to use the PID controller to regulate something. So the "Plant / process" from the following diagram from [wikipedia](https://en.wikipedia.org/wiki/PID_controller):
[image]

m-lundberg, 2022-09-25:
As explained in the comment above, the controlled system is whatever "thing" that you want to control using the PID controller.

As shown by @mrakitin in his PR, the example does run successfully in CI. If you have a specific problem with running the example, please provide some more information of what goes wrong and we can try to help.

closed 2022-09-25 by m-lundberg