m-lundberg/simple-pid #70: Improve tests

m-lundberg, 2022-10-04:
The test suite for this library could be improved. The biggest problem is that it's not completely deterministic, so sometimes tests can fail due to timing and pass if you just rerun it. It should be fixed so that you get the same result every time running a test. Possibly `dt` and/or `time_fn()` could help. Some care needs to be taken so that the tests still test what they intend to test though.

The tests also take a needlessly long time to run, mostly because of `test_converge_system()`. It would be nice if this could be sped up.

I'm sure one could think of more tests to add too.

gonzalo-bulnes, 2023-10-01:
> Possibly `dt` and/or `time_fn()` could help. 

I wrote https://github.com/gonzalo-bulnes/stime some time ago, mostly as an exercise... but, reading this issue left me thinking.

Is something like that what you had in mind when writing that `time_fn()` could help?

gonzalo-bulnes, 2023-10-02:
I took a look, I can help. I'll open a PR that addresses both:

- making tests deterministic
- making tests significantly faster