m-lundberg/simple-pid #89: import issue

NLuca, 2024-09-13:
I can not import the library.
import simple-pid as PID --> "SyntaxError: invalid syntax"
What do you recommend?
Thank you.

m-lundberg, 2024-09-14:
As the error suggests, the module could not be found. Please double check that you installed the library correctly with pip.

NLuca, 2024-09-16:
Hi Martin,

I am using PyCharm for my project. I have tried to install the module
through the PyCharm interface and also using the PyCharm terminal. I have
the same errors in both cases. What do you recommend?

Best regards,
Luca Neri

NLuca, 2024-09-16:
Hi Martin,

I have just solved the problem by installing the module outside PyCharm.

Thank you.
Best regards,
Luca Neri

closed 2024-09-17 by m-lundberg